Seeed-Studio/Digital_Infrared_Temperature_Sensor_MLX90615 #2: EEPROM emissivity change

plang, 2015-10-26:
Hi,

I'm playing with the emissivity which can be configured in the EEPROM of the MLX90615, and there is something strange when I try to modify it: it doesn't seem to change, unless I zero-out the configuration prior to setting the new value.

Here are the details: let's say the MLX90615 is configured with the defaults (1.0 emissivity): a call to `readEEPROM` shows in the console

```c
eepromData: 0x4000
```
Fine. Now let's try to set the emissivity to 0.98, with the following code:

```c
float emissivity = 0.98;
uint16_t emissivity_internal;

emissivity_internal = uint16_t(16384*emissivity);

mlx90615.writeEEPROM(emissivity_internal);
mlx90615.readEEPROM();
```
I'm expecting in this case an `eepromData: 0x3EB8`, but no, I keep on reading an `eepromData: 0x4000`, even if I call this code several times.

Things work as expected only if I zero-out the emissivity prior to trying to set it to 0.98 for example. This workaround apparently works fine:

```c
float emissivity = 0.98;
uint16_t emissivity_internal;

emissivity_internal = uint16_t(16384*emissivity);

mlx90615.writeEEPROM(0);
delay(500);
  
mlx90615.writeEEPROM(emissivity_internal); 
mlx90615.readEEPROM();
```
Note the call to `delay` which is necessary between two calls to `writeEEPROM`.

Do you have the same problem with other chips? I'm using your grove sensor.

plang, 2015-10-26:
Section 8.4.3.1.2 of the datasheet:
```
Note: Before a write operation takes place, the EEPROM cell needs to be erased. 
An erase operation is simply a write of 0000h at the same EEPROM address. 
Care needs to be taken not to alter factory calibration (EEPROM addresses 4…Dh).
````
Well, I think it's clear enough! :)

I just have to understand this timing problem that forces me to use a `delay(500)`after erasing the EEPROM.

Pillar1989, 2019-10-14:
I'm sorry that I didn't have time to deal with your problem. This issue is closed first. If you still need to discuss, you can reopen it, and we will try our best to help you

closed 2019-10-14 by Pillar1989