Seeed-Studio/Digital_Infrared_Temperature_Sensor_MLX90615 #4: Due support

Resonanz, 2016-07-18:
I am wondering about whether you could add support for the Arduino Due?  Presently it fails to compile for Due processors.

superyarik, 2016-10-19:
also request this, how define variables for this unknown CPU?

pushmeaway2, 2018-07-16:
Same here... i followed the seeedstudio tzutorial but get an #unknown cpu error.
Can anybody help please?

KillingJacky, 2018-08-06:
@pushmeaway2 the tutorial's instructing to use the soft i2c, why could you encounter the unknown CPU error? To have the I2cMaster library support Arduino Due, it's not just as simple as adding some varibles in I2cMaster.h, the interrupt mechanism is different from that of AVR chips, it seems that I2cMaster library has no plan to support Arduino Due since the Due's user volume. So please consider using the soft i2c.

lanselambor, 2018-08-15:
@pushmeaway2 has fixed Seeeduino 4.2  compilation error "error unknown CPU" by adding ARDUINO_ARCH_AVR  to the board library.

closed 2018-09-12 by KillingJacky